ros2/rclcpp #2796: async_send_goal deadlocks

robin-zealrobotics, 2025-04-09:
### Operating System:

Ubuntu 24.04

### ROS version or commit hash:

jazzy

### RMW implementation (if applicable):

rmw_cyclonedds_cpp (also on rmw_zenoh_cpp)

### Client library (if applicable):

rclcpp

### Steps to reproduce issue

Inconsistent

### Expected behavior

async_send_goal always returns a future without blocking

### Actual behavior

async_send_goal deadlocks

### Additional information

I don't know enough of the internals of action clients to know if this is a bug or not.
The setup is an action server and action client (part of many other nodes in my launch setup), both using the SingleThreadedExecutor. They happily work together and callbacks for the actions are received. But once in a while, the async_send_goal gets deadlocked. I still see the log statements of the action server completing the previous actions, but the corresponding log statements for the callbacks in the action client are never printed, as the SingleThreadedExecutor never gets to processing these as it gets stuck on the async_send_goal call.

I guess the first question is whether I'm reporting a bug or whether this is just me not understanding what should happen. Should it by design be theoretically impossible for an async_send_goal to deadlock when using a SingleThreadedExecutor? That was my understanding, but I might be wrong on that.

Deadlocked stack trace:
```
libc.so.6!futex_wait(int private, unsigned int expected, unsigned int * futex_word) (futex-internal.h:146)
libc.so.6!__GI___lll_lock_wait(int * futex, int * futex@entry, int private) (lowlevellock.c:49)
libc.so.6!lll_mutex_lock_optimized(pthread_mutex_t * mutex) (pthread_mutex_lock.c:48)
libc.so.6!___pthread_mutex_lock(pthread_mutex_t * mutex) (pthread_mutex_lock.c:93)
librclcpp_action.so!rclcpp_action::ClientBase::send_goal_request(std::shared_ptr<void>, std::function<void (std::shared_ptr<void>)>) (Unknown Source:0)
rclcpp_action::Client<zeal_msgs::action::Move>::async_send_goal(class rclcpp_action::Client<zeal_msgs::action::Move> * const this, const rclcpp_action::Client<zeal_msgs::action::Move>::Goal & goal, const struct rclcpp_action::Client<zeal_msgs::action::Move>::SendGoalOptions & options) (/opt/ros/jazzy/include/rclcpp_action/rclcpp_action/client.hpp:441)
ZealPlanner::NavigateToPoseActionServer::processQueue(class ZealPlanner::NavigateToPoseActionServer * const this) (/ament/overlay_ws/src/zeal_picker/zeal_planner/src/navigate_to_pose_action_server.cpp:257)
ZealPlanner::NavigateToPoseActionServer::processQueue()::{lambda(std::shared_ptr<rclcpp_action::ClientGoalHandle<zeal_msgs::action::Move> > const&)#1}::operator()(std::shared_ptr<rclcpp_action::ClientGoalHandle<zeal_msgs::action::Move> > const&) const(const struct {...} * const __closure, const rclcpp_action::ClientGoalHandle<zeal_msgs::action::Move>::SharedPtr & handle) (/ament/overlay_ws/src/zeal_picker/zeal_planner/src/navigate_to_pose_action_server.cpp:201)
std::__invoke_impl<void, ZealPlanner::NavigateToPoseActionServer::processQueue()::{lambda(std::shared_ptr<rclcpp_action::ClientGoalHandle<zeal_msgs::action::Move> > const&)#1}&, std::shared_ptr<rclcpp_action::ClientGoalHandle<zeal_msgs::action::Move> > >(std::__invoke_other, ZealPlanner::NavigateToPoseActionServer::processQueue()::{lambda(std::shared_ptr<rclcpp_action::ClientGoalHandle<zeal_msgs::action::Move> > const&)#1}&, std::shared_ptr<rclcpp_action::ClientGoalHandle<zeal_msgs::action::Move> >&&)(struct {...} & __f) (/usr/include/c++/13/bits/invoke.h:61)
std::__invoke_r<void, ZealPlanner::NavigateToPoseActionServer::processQueue()::{lambda(std::shared_ptr<rclcpp_action::ClientGoalHandle<zeal_msgs::action::Move> > const&)#1}&, std::shared_ptr<rclcpp_action::ClientGoalHandle<zeal_msgs::action::Move> > >(ZealPlanner::NavigateToPoseActionServer::processQueue()::{lambda(std::shared_ptr<rclcpp_action::ClientGoalHandle<zeal_msgs::action::Move> > const&)#1}&, std::shared_ptr<rclcpp_action::ClientGoalHandle<zeal_msgs::action::Move> >&&)(struct {...} & __fn) (/usr/include/c++/13/bits/invoke.h:111)
std::_Function_handler<void (std::shared_ptr<rclcpp_action::ClientGoalHandle<zeal_msgs::action::Move> >), ZealPlanner::NavigateToPoseActionServer::processQueue()::{lambda(std::shared_ptr<rclcpp_action::ClientGoalHandle<zeal_msgs::action::Move> > const&)#1}>::_M_invoke(std::_Any_data const&, std::shared_ptr<rclcpp_action::ClientGoalHandle<zeal_msgs::action::Move> >&&)(const union std::_Any_data & __functor, class std::shared_ptr<rclcpp_action::ClientGoalHandle<zeal_msgs::action::Move> > && __args#0) (/usr/include/c++/13/bits/std_function.h:290)
std::function<void (std::shared_ptr<rclcpp_action::ClientGoalHandle<zeal_msgs::action::Move> >)>::operator()(std::shared_ptr<rclcpp_action::ClientGoalHandle<zeal_msgs::action::Move> >) const(const class std::function<void(std::shared_ptr<rclcpp_action::ClientGoalHandle<zeal_msgs::action::Move> >)> * const this, class std::shared_ptr<rclcpp_action::ClientGoalHandle<zeal_msgs::action::Move> > __args#0) (/usr/include/c++/13/bits/std_function.h:591)
rclcpp_action::Client<zeal_msgs::action::Move>::async_send_goal(zeal_msgs::action::Move_Goal_<std::allocator<void> > const&, rclcpp_action::Client<zeal_msgs::action::Move>::SendGoalOptions const&)::{lambda(std::shared_ptr<void>)#1}::operator()(std::shared_ptr<void>)(struct {...} * const __closure, class std::shared_ptr<void> response) (/opt/ros/jazzy/include/rclcpp_action/rclcpp_action/client.hpp:466)
std::__invoke_impl<void, rclcpp_action::Client<zeal_msgs::action::Move>::async_send_goal(zeal_msgs::action::Move_Goal_<std::allocator<void> > const&, rclcpp_action::Client<zeal_msgs::action::Move>::SendGoalOptions const&)::{lambda(std::shared_ptr<void>)#1}&, std::shared_ptr<void> >(std::__invoke_other, rclcpp_action::Client<zeal_msgs::action::Move>::async_send_goal(zeal_msgs::action::Move_Goal_<std::allocator<void> > const&, rclcpp_action::Client<zeal_msgs::action::Move>::SendGoalOptions const&)::{lambda(std::shared_ptr<void>)#1}&, std::shared_ptr<void>&&)(struct {...} & __f) (/usr/include/c++/13/bits/invoke.h:61)
std::__invoke_r<void, rclcpp_action::Client<zeal_msgs::action::Move>::async_send_goal(zeal_msgs::action::Move_Goal_<std::allocator<void> > const&, rclcpp_action::Client<zeal_msgs::action::Move>::SendGoalOptions const&)::{lambda(std::shared_ptr<void>)#1}&, std::shared_ptr<void> >(rclcpp_action::Client<zeal_msgs::action::Move>::async_send_goal(zeal_msgs::action::Move_Goal_<std::allocator<void> > const&, rclcpp_action::Client<zeal_msgs::action::Move>::SendGoalOptions const&)::{lambda(std::shared_ptr<void>)#1}&, std::shared_ptr<void>&&)(struct {...} & __fn) (/usr/include/c++/13/bits/invoke.h:111)
std::_Function_handler<void (std::shared_ptr<void>), rclcpp_action::Client<zeal_msgs::action::Move>::async_send_goal(zeal_msgs::action::Move_Goal_<std::allocator<void> > const&, rclcpp_action::Client<zeal_msgs::action::Move>::SendGoalOptions const&)::{lambda(std::shared_ptr<void>)#1}>::_M_invoke(std::_Any_data const&, std::shared_ptr<void>&&)(const union std::_Any_data & __functor, class std::shared_ptr<void> && __args#0) (/usr/include/c++/13/bits/std_function.h:290)
librclcpp_action.so!rclcpp_action::ClientBase::handle_goal_response(rmw_request_id_s const&, std::shared_ptr<void>) (Unknown Source:0)
librclcpp_action.so!rclcpp_action::ClientBase::execute(std::shared_ptr<void> const&) (Unknown Source:0)
librclcpp.so!rclcpp::Executor::execute_any_executable(rclcpp::AnyExecutable&) (Unknown Source:0)
librclcpp.so!rclcpp::executors::SingleThreadedExecutor::spin() (Unknown Source:0)
```

robin-zealrobotics, 2025-04-09:
Ah, is it fundamentally impossible to do an async_send_goal call from within another action's goal_response_callback? (in SingleThreadedExecutor)
Because my code (sometimes) sends another action from within such a callback (which is also what's happening in the stacktrace), so that would explain that I'm not always seeing it deadlock.

robin-zealrobotics, 2025-04-09:
I assume the problem I'm encountering is something similar to this and it's not a bug, but just a mistake on my end.
https://robotics.stackexchange.com/questions/45241/is-it-possible-to-call-a-ros-service-from-another-services-callback-function-in

If so, feel free to close this issue.

fujitatomoya, 2025-04-09:
> Because my code (sometimes) sends another action from within such a callback (which is also what's happening in the stacktrace), so that would explain that I'm not always seeing it deadlock.

you can call async API in the callback, but you cannot wait on that in the callback.

https://docs.ros.org/en/rolling/How-To-Guides/Sync-Vs-Async.html

robin-zealrobotics, 2025-04-09:
> you can call async API in the callback, but you cannot wait on that in the callback.
>
> https://docs.ros.org/en/rolling/How-To-Guides/Sync-Vs-Async.html

Well no, that’s not what I’m doing, I’m not calling wait on the future, I’m not even able to get a future. If you look at my stack trace it’s blocked on the async send goal. 
So from what you are saying, it might actually be a bug then, if by design it should work. 

(Side note, in my code I’ve now changed it to deferring it with a one shot timer and I can’t reproduce the problem anymore)

fujitatomoya, 2025-04-09:
> If you look at my stack trace it’s blocked on the async send goal.

ah, sorry. now i see that is the self-deadlock on

https://github.com/ros2/rclcpp/blob/6040228d13547d44d373ce3f4f15afa6c8347232/rclcpp_action/src/client_base.cpp#L174-L175

i think we can make your case work with using recursive mutex lock, but i am not sure if this `sending action request on goal response callback` design is recommended. feel free to reopen this to get more opinion if you want!

robin-zealrobotics, 2025-04-09:
My case was that I have an action server that supports a FIFO queue of actions. The action client wants to similarly schedule the actions FIFO. In order to avoid actions getting out-of-order, my first thought was to ensure the previous action was accepted. Hence, I allow the next action to be scheduled in the goal response callback, as then the order should be guaranteed and it's the earliest time I can schedule the next action.

Maybe it's good to reopen the issue in case others think this is something that should be supported. But it's not a necessity for me, as I'm now working with one-shot timers.
However, given that async calls are documented as the best way to avoid deadlocks (which they are, aside from the obvious 'wait' deadlocks), maybe we do indeed want to use a recursive mutex here.

(Btw, I can't reopen it)

fujitatomoya, 2025-04-09:
thanks for sharing the use case that you have. i will reopen and have discussion on this more.

fujitatomoya, 2025-04-09:
@robin-zealrobotics just FYI, https://github.com/ros2/rclcpp/pull/2798 is opened.

fujitatomoya, 2025-04-11:
@robin-zealrobotics we actually fixed this with recursive mutex, but it is not going to be backported to released distro. (i am not inclined to do that since that is kind of behavior change) that said, your original queue mechanism works with current rolling head. i just wanted to let you know this since you created this issue. and thank you for posting the issue.

closed 2025-04-11 by fujitatomoya