ros2/rclcpp #2799: Performance regression in EventsExecutor on jazzy

jplapp, 2025-04-10:
### Operating System:

ubuntu 22

### ROS version or commit hash:

jazzy

### RMW implementation (if applicable):

rmw_cyclonedds_cpp and rmw_fastdds_cpp

### Client library (if applicable):

rclcpp

### Steps to reproduce issue

1. Run any action server with EventsExecutor, e.g. [from ros2 demos](https://github.com/ros2/demos/tree/rolling/action_tutorials/action_tutorials_cpp) , compile with EventsExecutor specified `rclcpp_components_register_node(action_tutorials PLUGIN "action_tutorials_cpp::FibonacciActionServer" EXECUTABLE fibonacci_action_server EXECUTOR EventsExecutor)` 
2. `ros2 run action_tutorials_cpp fibonacci_action_server` 
3. Check CPU load 

### Expected behavior

EventsExecutor has similar or less CPU load than SingleThreadedExecutor when running an idle action server

Running with SingleThreadedExecutor the CPU load is 0% (as nothing is happening)

### Actual behavior

CPU load is significantly more than 0, e.g. 50%: 

[image]

### Additional information

When I revert https://github.com/ros2/rclcpp/pull/2674 , it behaves as expected

MCFurry, 2025-04-10:
Ahh! We justed started a deep-dive and came to the same conclusion...

jmachowinski, 2025-04-10:
I'll have a look

jmachowinski, 2025-04-10:
@MCFurry  @jplapp 
Please check an verify that https://github.com/ros2/rclcpp/pull/2800 fixes the bug for you.

MCFurry, 2025-04-10:
@jmachowinski I just ran a very quick test from your branch, but unfortunately the issue is still there... Thanks for the quick effort though!

jmachowinski, 2025-04-10:
> [@jmachowinski](https://github.com/jmachowinski) I just ran a very quick test from your branch, but unfortunately the issue is still there... Thanks for the quick effort though!

How exactly did you test ? I checked with the procedure described in the first post, and the issue is solved for me using the patch. I just rechecked it on a second PC.

jplapp, 2025-04-10:
From my side I can verify that the issue is resolved, both in the test case as well as the full system.

@jmachowinski thanks for the quick fix!

closed 2025-04-10 by jplapp

MCFurry, 2025-04-10:
Hmm my bad, it seems during my test only rclcpp was build from source but rclcpp_action was not. Now testing again and the issue indeed seems solved. Thanks again for the quick fix!