ros2/rclcpp #2803: EventsExecutor misses one-shot-timer

robin-zealrobotics, 2025-04-11:
### Operating System:

```Linux robin-zeal 6.9.3-76060903-generic #202405300957~1732141768~22.04~f2697e1~dev-Ubuntu SMP PREEMPT_DY x86_64 x86_64 x86_64 GNU/Linux```

### ROS version or commit hash:

jazzy

### RMW implementation (if applicable):

rmw_zenoh_cpp

### Client library (if applicable):

rclcpp

### 'ros2 doctor --report' output

<details><summary>ros2 doc --report</summary>

```console
<COPY OUTPUT HERE>
```
</details>

### Steps to reproduce issue

Run this code
```
#include <chrono>
#include <memory>

#include <rclcpp/rclcpp.hpp>

using namespace std::chrono_literals;

class ExecutorBugNode : public rclcpp::Node
{
public:
  ExecutorBugNode()
  : Node("executor_bug_node")
  {
    wall_timer_ = create_wall_timer(
      50ms, std::bind(&ExecutorBugNode::wall_timer_callback, this));
  }

private:
  void
  wall_timer_callback()
  {
    if (counter_ < 100) {
      std::this_thread::sleep_for(40ms);
      counter_++;
    } else {
      RCLCPP_INFO(get_logger(), "Sleep stopped");
    }

    if (!one_shot_timer_) {
      one_shot_timer_ = create_wall_timer(
        0s, std::bind(&ExecutorBugNode::one_shot_timer_callback, this));
    } else {
      RCLCPP_INFO(get_logger(), "One-shot timer already exists. Skipping creation.");
    }
  }

  void
  one_shot_timer_callback()
  {
    RCLCPP_INFO(get_logger(), "--- One-shot timer callback triggered! ---");
    if (one_shot_timer_) {
[... 156 lines cut ...]
[INFO] [2025-04-11 07:26:09.556] [executor_bug_node]: One-shot timer already exists. Skipping creation.
[INFO] [2025-04-11 07:26:09.605] [executor_bug_node]: Sleep stopped
[INFO] [2025-04-11 07:26:09.606] [executor_bug_node]: One-shot timer already exists. Skipping creation.
[INFO] [2025-04-11 07:26:09.655] [executor_bug_node]: Sleep stopped
[INFO] [2025-04-11 07:26:09.656] [executor_bug_node]: One-shot timer already exists. Skipping creation.
[INFO] [2025-04-11 07:26:09.705] [executor_bug_node]: Sleep stopped
[INFO] [2025-04-11 07:26:09.706] [executor_bug_node]: One-shot timer already exists. Skipping creation.
[INFO] [2025-04-11 07:26:09.755] [executor_bug_node]: Sleep stopped
[INFO] [2025-04-11 07:26:09.756] [executor_bug_node]: One-shot timer already exists. Skipping creation.
[INFO] [2025-04-11 07:26:09.805] [executor_bug_node]: Sleep stopped
[INFO] [2025-04-11 07:26:09.806] [executor_bug_node]: One-shot timer already exists. Skipping creation.
[INFO] [2025-04-11 07:26:09.855] [executor_bug_node]: Sleep stopped
[INFO] [2025-04-11 07:26:09.856] [executor_bug_node]: One-shot timer already exists. Skipping creation.
[INFO] [2025-04-11 07:26:09.906] [executor_bug_node]: Sleep stopped
[INFO] [2025-04-11 07:26:09.906] [executor_bug_node]: One-shot timer already exists. Skipping creation.
[INFO] [2025-04-11 07:26:09.955] [executor_bug_node]: Sleep stopped
[INFO] [2025-04-11 07:26:09.955] [executor_bug_node]: One-shot timer already exists. Skipping creation.
[INFO] [2025-04-11 07:26:10.589] [executor_bug_node]: Sleep stopped
[INFO] [2025-04-11 07:26:10.601] [executor_bug_node]: One-shot timer already exists. Skipping creation.
[INFO] [2025-04-11 07:26:10.558] [executor_bug_node]: Sleep stopped
[INFO] [2025-04-11 07:26:10.559] [executor_bug_node]: One-shot timer already exists. Skipping creation.
[INFO] [2025-04-11 07:26:10.105] [executor_bug_node]: Sleep stopped
[INFO] [2025-04-11 07:26:10.106] [executor_bug_node]: One-shot timer already exists. Skipping creation.
....
```

fujitatomoya, 2025-04-11:
@robin-zealrobotics @jmachowinski just checking, this problem happens on rolling too, right?

jmachowinski, 2025-04-11:
Yes