ros2/rclcpp #2807: SimpleEventsQueue::enqueue breaks event semantics by altering num_events and duplicating the event

zhihaoshang, 2025-04-13:
### Operating System:

```Linux shangzh-VMware-Virtual-Platform 6.11.0-21-generic #21~24.04.1-Ubuntu SMP PREEMPT_DYNAMIC Mon Feb 24 16:52:15 UTC 2 x86_64 x86_64 x86_64 GNU/Linux```

### ROS version or commit hash:

ros2 jazzy

### Client library (if applicable):

rclcpp

### 'ros2 doctor --report' output

<details><summary>ros2 doc --report</summary>

```console
  NETWORK CONFIGURATION
inet         : 127.0.0.1
inet4        : ['127.0.0.1']
inet6        : ['::1']
netmask      : 255.0.0.0
device       : lo
flags        : 73<LOOPBACK,RUNNING,UP>
mtu          : 65536
inet         : 192.168.148.137
inet4        : ['192.168.148.137']
ether        : 00:0c:29:be:c8:19
inet6        : ['fe80::20c:29ff:febe:c819%ens33']
netmask      : 255.255.255.0
device       : ens33
flags        : 4163<MULTICAST,BROADCAST,RUNNING,UP>
mtu          : 1500
broadcast    : 192.168.148.255

   PACKAGE VERSIONS
rqt_bag_plugins                           : latest=1.5.4, local=1.5.4
rosbag2                                   : latest=0.26.6, local=0.26.6
rosbag2_storage_default_plugins           : latest=0.26.6, local=0.26.6
rosbag2_storage_mcap                      : latest=0.26.6, local=0.26.6
rosbag2_compression_zstd                  : latest=0.26.6, local=0.26.6
mcap_vendor                               : latest=0.26.6, local=0.26.6
zstd_vendor                               : latest=0.26.6, local=0.26.6
rviz_visual_testing_framework             : latest=14.1.8, local=14.1.7
rviz2                                     : latest=14.1.8, local=14.1.7
rviz_default_plugins                      : latest=14.1.8, local=14.1.7
rviz_common                               : latest=14.1.8, local=14.1.7
rqt_bag                                   : latest=1.5.4, local=1.5.4
rosbag2_examples_py                       : latest=0.26.6, local=0.26.6
launch_testing_examples                   : latest=0.19.5, local=0.19.4
ros2bag                                   : latest=0.26.6, local=0.26.6
rosbag2_py                                : latest=0.26.6, local=0.26.6
rosbag2_transport                         : latest=0.26.6, local=0.26.6
rosbag2_storage_sqlite3                   : latest=0.26.6, local=0.26.6
rosbag2_examples_cpp                      : latest=0.26.6, local=0.26.6
rosbag2_compression                       : latest=0.26.6, local=0.26.6
rosbag2_cpp                               : latest=0.26.6, local=0.26.6
rosbag2_storage                           : latest=0.26.6, local=0.26.6
image_common                              : latest=5.1.6, local=5.1.5
camera_info_manager                       : latest=5.1.6, local=5.1.5
camera_calibration_parsers                : latest=5.1.6, local=5.1.5
yaml_cpp_vendor                           : latest=9.0.1, local=9.0.1
interactive_markers                       : latest=2.5.4, local=2.5.4
common_interfaces                         : latest=5.3.6, local=5.3.5
visualization_msgs                        : latest=5.3.6, local=5.3.5
dummy_robot_bringup                       : latest=0.33.5, local=0.33.5
robot_state_publisher                     : latest=3.3.3, local=3.3.3
kdl_parser                                : latest=2.11.0, local=2.11.0
urdf                                      : latest=2.10.0, local=2.10.0
urdf_parser_plugin                        : latest=2.10.0, local=2.10.0
[... 491 lines cut ...]
dequeue() should return that exact event;

the queue size should reflect one logical event, i.e., size() == 1.

### Actual behavior

enqueue overwrites event.num_events with 1;

and pushes the event into the queue 1,000,000 times;

resulting in a dequeued event with num_events = 1;

and a queue size of 999,999 after dequeue;

### Additional information

## Suggested Fix
Update the SimpleEventsQueue::enqueue logic to retain the integrity of the event. It should not modify the num_events field or duplicate the event internally. The queue should treat each ExecutorEvent as an atomic unit.
```cpp
void enqueue(const ExecutorEvent & event) override {
  std::unique_lock<std::mutex> lock(mutex_);
  event_queue_.push(event);  // No modification or duplication
  events_queue_cv_.notify_one();
}
```

zhihaoshang, 2025-04-13:
## Minimize the instance

Test Case
```cpp
#include <gtest/gtest.h>
#include <memory>
#include "rclcpp/experimental/executors/events_executor/events_executor_event_types.hpp"
#include "rclcpp/experimental/executors/events_executor/simple_events_queue.hpp"

using namespace std::chrono_literals;

TEST(TestEventsQueue, StressTestWithHighNumEvents) {
  auto simple_queue = std::make_unique<rclcpp::experimental::executors::SimpleEventsQueue>();
  rclcpp::experimental::executors::ExecutorEvent large_event {};
  large_event.num_events = 1000000;
  
  EXPECT_TRUE(simple_queue->empty());

  simple_queue->enqueue(large_event);
  EXPECT_FALSE(simple_queue->empty());
  EXPECT_EQ(simple_queue->size(), 1u);

  rclcpp::experimental::executors::ExecutorEvent dequeued_event {};
  bool ret = simple_queue->dequeue(dequeued_event);

  EXPECT_TRUE(ret);
  EXPECT_EQ(dequeued_event.num_events, large_event.num_events);
  EXPECT_TRUE(simple_queue->empty());
  EXPECT_EQ(simple_queue->size(), 0u);
}
```

Output
```markdown
[==========] Running 1 test from 1 test suite.
[----------] Global test environment set-up.
[----------] 1 test from TestEventsQueue
[ RUN      ] TestEventsQueue.StressTestWithHighNumEvents
/home/shangzh/ros2_jazzy/src/ros2/rclcpp/rclcpp/test/rclcpp/executors/test_events_queue.cpp:17: Failure
Expected equality of these values:
  simple_queue->size()
    Which is: 1000000
  1u
    Which is: 1

/home/shangzh/ros2_jazzy/src/ros2/rclcpp/rclcpp/test/rclcpp/executors/test_events_queue.cpp:23: Failure
Expected equality of these values:
  dequeued_event.num_events
    Which is: 1
  large_event.num_events
    Which is: 1000000

/home/shangzh/ros2_jazzy/src/ros2/rclcpp/rclcpp/test/rclcpp/executors/test_events_queue.cpp:24: Failure
Value of: simple_queue->empty()
  Actual: false
Expected: true

/home/shangzh/ros2_jazzy/src/ros2/rclcpp/rclcpp/test/rclcpp/executors/test_events_queue.cpp:25: Failure
Expected equality of these values:
  simple_queue->size()
    Which is: 999999
  0u
    Which is: 0

[  FAILED  ] TestEventsQueue.StressTestWithHighNumEvents (76 ms)
[----------] 1 test from TestEventsQueue (77 ms total)

[----------] Global test environment tear-down
[==========] 1 test from 1 test suite ran. (77 ms total)
[  PASSED  ] 0 tests.
[  FAILED  ] 1 test, listed below:
[  FAILED  ] TestEventsQueue.StressTestWithHighNumEvents

 1 FAILED TEST
```

alsora, 2025-04-13:
Hi @zhihaoshang,
I agree with your analysis, but I do have some doubts about whether we should make any change.

The Events Queue class is an internal implementation detail of the events executor; it shouldn't be used outside of it.
You say that:
> The enqueue method should preserve the original event object, including its num_events field, and enqueue it as-is. Upon dequeue, the event should retain the original num_events value.

But why is this really expected besides the fact that a normal `std::queue` behaves like that?

The only way to push events into the queue is via APIs like `rmw_event_callback_t`.
How the data passed to the callback maps to events is again a private implementation detail of the events executor.

The `num_events` field is just a convenience to allows for the existance of more advanced types of queues.
However the SimpleEventQueue requires that each element represents a single event.
For this reason if a user tries to enqueue an event with `num_events > 1` this gets automatically split among multiple elements.

Note that dequeuing events with `num_events > 1` would be problematic.
We would need extra logic in the executor class because if we just did
```
for (int i = 0; i < num_events; i++)
{
     execute_subscription();
}
```
we could theoretically block for a very long time.

What we could do is to improve the comments around the simple queue class.

fujitatomoya, 2025-04-13:
i agree with @alsora , for me this does not look like a bug. (maybe we can add some more docstring and test cases though.)

@zhihaoshang how do you find this is the bug?

zhihaoshang, 2025-04-15:
Hi @alsora @fujitatomoya, thank you both for the detailed response.

I understand and respect your perspective — especially that SimpleEventsQueue is an internal component, and that its current design choice favors simplicity by ensuring each enqueued element represents a single event. I also agree that modifying the documentation to clarify this behavior would be helpful.

That said, I'd like to reiterate the main concern: the current enqueue implementation modifies the input event object in-place, which may be surprising to users and leads to semantic inconsistency — especially when writing unit tests or debugging behavior.

Even if this queue is not intended for external use, ensuring self-consistent behavior and minimal side effects can still benefit long-term maintenance and developer understanding. One possible alternative could be to copy the input event and internally adjust its num_events field, rather than altering the original.

I fully understand if no functional change is made, but I would strongly suggest at least:

1.Documenting clearly that the enqueue method splits multi-event objects and modifies num_events;

2.And perhaps adding a safeguard or assertion to catch unexpected values early during enqueue.

Thanks again for your time and clarification!

fujitatomoya, 2025-04-15:
> One possible alternative could be to copy the input event and internally adjust its num_events field, rather than altering the original.

this does not make sense, because this is already happening.

@zhihaoshang the same question with other issues that you opened. are you using GenAI or any static tools to create the issue? can you share what kind of tools are you using to create issue and response?

Reference: https://discourse.ros.org/t/interim-policy-on-the-use-of-generative-ai-in-osrf-projects/39731

zhihaoshang, 2025-04-16:
> > 一种可能的替代方法是复制 input 事件并在内部调整其 num_events 字段，而不是更改原始字段。
> 
> 这没有意义，因为这种情况已经发生了。
> 
> 与您打开的其他问题相同的问题。您是否使用 GenAI 或任何静态工具来创建问题？您能否分享一下您使用了哪些类型的工具来创建 Issue 和 Response？
> 
> 编号： https://discourse.ros.org/t/interim-policy-on-the-use-of-generative-ai-in-osrf-projects/39731

Hi, thank you for your reply! I used some auxiliary tools when writing the responses, currently using ChatGPT, to help improve the clarity of expression. Please feel free to let me know if there's anything inappropriate.

closed 2025-04-16 by zhihaoshang