ros2/rclcpp #2809: Missing Parameter Consistency Check in ParameterEventsFilter Constructor

zhihaoshang, 2025-04-14:
### Operating System:

```Linux shangzh-VMware-Virtual-Platform 6.11.0-21-generic #21~24.04.1-Ubuntu SMP PREEMPT_DYNAMIC Mon Feb 24 16:52:15 UTC 2 x86_64 x86_64 x86_64 GNU/Linux```

### ROS version or commit hash:

ros2 jazzy

### Client library (if applicable):

rclcpp

### 'ros2 doctor --report' output

<details><summary>ros2 doc --report</summary>

```console
   NETWORK CONFIGURATION
inet         : 127.0.0.1
inet4        : ['127.0.0.1']
inet6        : ['::1']
netmask      : 255.0.0.0
device       : lo
flags        : 73<UP,RUNNING,LOOPBACK>
mtu          : 65536
inet         : 192.168.148.137
inet4        : ['192.168.148.137']
ether        : 00:0c:29:be:c8:19
inet6        : ['fe80::20c:29ff:febe:c819%ens33']
netmask      : 255.255.255.0
device       : ens33
flags        : 4163<MULTICAST,UP,BROADCAST,RUNNING>
mtu          : 1500
broadcast    : 192.168.148.255

   PACKAGE VERSIONS
rqt_bag_plugins                           : latest=1.5.4, local=1.5.4
rosbag2                                   : latest=0.26.6, local=0.26.6
rosbag2_storage_default_plugins           : latest=0.26.6, local=0.26.6
rosbag2_storage_mcap                      : latest=0.26.6, local=0.26.6
rosbag2_compression_zstd                  : latest=0.26.6, local=0.26.6
mcap_vendor                               : latest=0.26.6, local=0.26.6
zstd_vendor                               : latest=0.26.6, local=0.26.6
rviz_visual_testing_framework             : latest=14.1.8, local=14.1.7
rviz2                                     : latest=14.1.8, local=14.1.7
rviz_default_plugins                      : latest=14.1.8, local=14.1.7
rviz_common                               : latest=14.1.8, local=14.1.7
rqt_bag                                   : latest=1.5.4, local=1.5.4
rosbag2_examples_py                       : latest=0.26.6, local=0.26.6
launch_testing_examples                   : latest=0.19.5, local=0.19.4
ros2bag                                   : latest=0.26.6, local=0.26.6
rosbag2_py                                : latest=0.26.6, local=0.26.6
rosbag2_transport                         : latest=0.26.6, local=0.26.6
rosbag2_storage_sqlite3                   : latest=0.26.6, local=0.26.6
rosbag2_examples_cpp                      : latest=0.26.6, local=0.26.6
rosbag2_compression                       : latest=0.26.6, local=0.26.6
rosbag2_cpp                               : latest=0.26.6, local=0.26.6
rosbag2_storage                           : latest=0.26.6, local=0.26.6
image_common                              : latest=5.1.6, local=5.1.5
camera_info_manager                       : latest=5.1.6, local=5.1.5
camera_calibration_parsers                : latest=5.1.6, local=5.1.5
yaml_cpp_vendor                           : latest=9.0.1, local=9.0.1
interactive_markers                       : latest=2.5.4, local=2.5.4
common_interfaces                         : latest=5.3.6, local=5.3.5
visualization_msgs                        : latest=5.3.6, local=5.3.5
dummy_robot_bringup                       : latest=0.33.5, local=0.33.5
robot_state_publisher                     : latest=3.3.3, local=3.3.3
kdl_parser                                : latest=2.11.0, local=2.11.0
urdf                                      : latest=2.10.0, local=2.10.0
urdf_parser_plugin                        : latest=2.10.0, local=2.10.0
[... 465 lines cut ...]

 1 FAILED TEST
```

### Expected behavior

When the names and types vectors have different lengths, the constructor should immediately throw a std::invalid_argument exception to prevent potential mismatches or undefined behavior. This is the reasonable expectation for the interface user and is reflected in the existing test case.

### Actual behavior

Currently, the ParameterEventsFilter constructor does not perform a length consistency check between names and types. When the lengths differ, it proceeds with the filtering logic, causing the unit test to fail because the expected exception is not thrown:
```cpp
Expected: rclcpp::ParameterEventsFilter(full, {"new"}, {nt, dt}) throws an exception of type std::invalid_argument.
Actual: it throws nothing.
```

### Additional information

##  Suggested Fix
It is recommended to add a parameter consistency check at the beginning of the constructor:
```cpp
if (!names.empty() && !types.empty() && names.size() != types.size()) {
  throw std::invalid_argument("Parameter names and types must match in size.");
}
```

fujitatomoya, 2025-04-14:
> When the names and types vectors have different lengths, the constructor should immediately throw a std::invalid_argument exception to prevent potential mismatches or undefined behavior. This is the reasonable expectation for the interface user and is reflected in the existing test case.

can you explain why names and types length needs to be the same? the same parameter name could be newly created and updated, even deleted. IMO, names and types are irrelevant.

zhihaoshang, 2025-04-15:
> > 当 names 和 types 向量具有不同的长度时，构造函数应立即引发 std：：invalid_argument 异常，以防止潜在的不匹配或未定义的行为。这是接口用户的合理期望，并反映在现有测试用例中。
> 
> 您能解释一下为什么名称和类型 length 需要相同吗？可以新创建和更新相同的参数名称，甚至可以删除相同的参数名称。IMO，名称和类型无关紧要。

Thank you for your response.

You're absolutely right that a single parameter name can appear in multiple event types (e.g., a parameter could be created and later updated). However, the constructor of ParameterEventsFilter currently takes two parallel vectors — names and types — and applies filtering based on matching pairs at the same index position. This design implies a positional correspondence between each name and its associated event type.

For example, the intention behind this call:
```cpp
rclcpp::ParameterEventsFilter(event, {"param1", "param2"}, {NEW, CHANGED});
```

is to filter for:

"param1" with event type NEW, and

"param2" with event type CHANGED.

If the two vectors are of unequal length, it introduces ambiguity: Should the filtering be Cartesian (all names × all types)? Should it truncate to the shorter length? Should it ignore the extra elements?

Currently, the constructor does not clarify this, and the implementation proceeds without validation, potentially filtering with mismatched or unintended (name, type) pairs — as reflected in the failed test case.

To avoid misuse and clarify intent, I believe the constructor should either:

1.Enforce equal lengths and validate at construction time (throwing an std::invalid_argument), or

2.Clearly document that the vectors are independent and the filter performs a Cartesian match across all names and all types.

If the second behavior is indeed intended (Cartesian product matching), I would suggest updating the interface or adding a new constructor to reflect that usage more explicitly, since the current API implies a paired relationship.

Thanks again for the discussion

fujitatomoya, 2025-04-15:
> However, the constructor of ParameterEventsFilter currently takes two parallel vectors — names and types — and applies filtering based on matching pairs at the same index position. This design implies a positional correspondence between each name and its associated event type.

I think this assumption is incorrect.

fujitatomoya, 2025-04-15:
@zhihaoshang according to the history, are you using GenAI or any static tools to create the issue? can you share what kind of tools are you using to create issue and response?

[image]

Reference: https://discourse.ros.org/t/interim-policy-on-the-use-of-generative-ai-in-osrf-projects/39731

zhihaoshang, 2025-04-16:
> 根据历史记录，您是否使用 GenAI 或任何静态工具来制造问题？您能否分享一下您使用了哪些类型的工具来创建 Issue 和 Response？
> 
> <img alt="Image" width="645" src="https://private-user-images.githubusercontent.com/43395114/433952504-9a775ac5-3efb-4e65-93e2-88dd06b1d66d.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3NDQ3NzAxMTgsIm5iZiI6MTc0NDc2OTgxOCwicGF0aCI6Ii80MzM5NTExNC80MzM5NTI1MDQtOWE3NzVhYzUtM2VmYi00ZTY1LTkzZTItODhkZDA2YjFkNjZkLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNTA0MTYlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjUwNDE2VDAyMTY1OFomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWM4NGZhOTE3MzI1ZmZmNWM4YWNkNWNlNWNiZGJmNzhkMTdiMjM3ZGE0ODI2MmM2ZjRjMWE1NzQ3Zjk1NzZkMDAmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.41xDCjoBQFtrNNiq-4Kxf_C1ZgW-St2oHq7_92DlTus">
> 编号： https://discourse.ros.org/t/interim-policy-on-the-use-of-generative-ai-in-osrf-projects/39731

Hi, thank you for your reply!I used some aids when writing this response, to help improve the clarity of expression. I wasn’t aware that this might be an issue, but I appreciate you pointing it out. Please feel free to let me know if any adjustments are needed.

closed 2025-04-16 by zhihaoshang