ros2/rclcpp #2810: Crash due to Unhandled Null Pointer in ParameterEventsFilter Constructor

zhihaoshang, 2025-04-14:
### Operating System:

```Linux shangzh-VMware-Virtual-Platform 6.11.0-21-generic #21~24.04.1-Ubuntu SMP PREEMPT_DYNAMIC Mon Feb 24 16:52:15 UTC 2 x86_64 x86_64 x86_64 GNU/Linux```

### ROS version or commit hash:

ros2 jazzy

### Client library (if applicable):

rclcpp

### 'ros2 doctor --report' output

<details><summary>ros2 doc --report</summary>

```console
   NETWORK CONFIGURATION
inet         : 127.0.0.1
inet4        : ['127.0.0.1']
inet6        : ['::1']
netmask      : 255.0.0.0
device       : lo
flags        : 73<RUNNING,UP,LOOPBACK>
mtu          : 65536
inet         : 192.168.148.137
inet4        : ['192.168.148.137']
ether        : 00:0c:29:be:c8:19
inet6        : ['fe80::20c:29ff:febe:c819%ens33']
netmask      : 255.255.255.0
device       : ens33
flags        : 4163<RUNNING,UP,BROADCAST,MULTICAST>
mtu          : 1500
broadcast    : 192.168.148.255

   PLATFORM INFORMATION
system           : Linux
platform info    : Linux-6.11.0-21-generic-x86_64-with-glibc2.39
release          : 6.11.0-21-generic
processor        : x86_64

   QOS COMPATIBILITY LIST
compatibility status    : No publisher/subscriber pairs found

   RMW MIDDLEWARE
middleware name    : rmw_fastrtps_cpp

   TOPIC LIST
topic               : none
publisher count     : 0
subscriber count    : 0

```
</details>

### Steps to reproduce issue

## Environment
OS Version: Ubuntu 24.04
rclcpp version: ros2 jazzy
Compiler name and version number: Ubuntu clang version 18.1.3
Source or binary build?
source build
build options: --mixin asan-gcc

## Description
When using the ParameterEventsFilter constructor, passing a null pointer (nullptr) causes the program to crash. This issue occurs because the constructor does not check for null pointers, leading to a crash when attempting to dereference the null pointer.

## Test Case
```cpp
[... 85 lines cut ...]
    #16 0x79bdeae2a28a in __libc_start_main_impl ../csu/libc-start.c:360
    #17 0x6020b485a0b4 in _start (/home/shangzh/ros2_jazzy/build/rclcpp/test/rclcpp/test_parameter_events_filter+0x960b4) (BuildId: 574b4520414cc61b5fc47f14a7898e117fe3790b)

AddressSanitizer can not provide additional info.
SUMMARY: AddressSanitizer: SEGV (/home/shangzh/ros2_jazzy/install/rclcpp/lib/librclcpp.so+0xda5bcb) (BuildId: 6fc5c95245c12254f84b0fb758e76bec499949f1) in __gnu_cxx::__normal_iterator<rcl_interfaces::msg::Parameter_<std::allocator<void> > const*, std::vector<rcl_interfaces::msg::Parameter_<std::allocator<void> >, std::allocator<rcl_interfaces::msg::Parameter_<std::allocator<void> > > > >::__normal_iterator(rcl_interfaces::msg::Parameter_<std::allocator<void> > const* const&)
==252400==ABORTING
```

### Expected behavior

When the ParameterEventsFilter constructor receives a null pointer (nullptr) as a parameter, it should throw a std::invalid_argument exception, rather than dereferencing the null pointer and causing a memory access error.

### Actual behavior

When a null pointer is passed, the ParameterEventsFilter constructor does not check for the null pointer, resulting in a crash (SEGV error) when attempting to dereference the null pointer during operations like std::find or accessing parameters.

### Additional information

## Suggested Fix
It is recommended to add a null pointer check at the beginning of the ParameterEventsFilter constructor. If the event parameter is null, throw a std::invalid_argument exception to prevent further memory access errors.
```cpp
if (!event) {
  throw std::invalid_argument("ParameterEvent cannot be null");
}
```

fujitatomoya, 2025-04-14:
this is a bug, it should handle the nullptr when `ParameterEventsFilter` object is constructed.

fujitatomoya, 2025-04-14:
@zhihaoshang i created https://github.com/ros2/rclcpp/pull/2814, it would be appreciated if you can review. thanks!

closed 2025-04-24 by fujitatomoya