ros2/rclcpp #2811: QoS::keep_last(0) does not throw an exception and silently falls back to SYSTEM_DEFAULT, potentially hiding user misconfiguration

zhihaoshang, 2025-04-14:
### Operating System:

```Linux shangzh-VMware-Virtual-Platform 6.11.0-21-generic #21~24.04.1-Ubuntu SMP PREEMPT_DYNAMIC Mon Feb 24 16:52:15 UTC 2 x86_64 x86_64 x86_64 GNU/Linux```

### ROS version or commit hash:

ros2 jazzy

### Client library (if applicable):

rclcpp

### 'ros2 doctor --report' output

<details><summary>ros2 doc --report</summary>

```console
   NETWORK CONFIGURATION
inet         : 127.0.0.1
inet4        : ['127.0.0.1']
inet6        : ['::1']
netmask      : 255.0.0.0
device       : lo
flags        : 73<RUNNING,UP,LOOPBACK>
mtu          : 65536
inet         : 192.168.148.137
inet4        : ['192.168.148.137']
ether        : 00:0c:29:be:c8:19
inet6        : ['fe80::20c:29ff:febe:c819%ens33']
netmask      : 255.255.255.0
device       : ens33
flags        : 4163<MULTICAST,BROADCAST,UP,RUNNING>
mtu          : 1500
broadcast    : 192.168.148.255

   PACKAGE VERSIONS
rqt_bag_plugins                           : latest=1.5.4, local=1.5.4
rosbag2                                   : latest=0.26.6, local=0.26.6
rosbag2_storage_default_plugins           : latest=0.26.6, local=0.26.6
rosbag2_storage_mcap                      : latest=0.26.6, local=0.26.6
rosbag2_compression_zstd                  : latest=0.26.6, local=0.26.6
mcap_vendor                               : latest=0.26.6, local=0.26.6
zstd_vendor                               : latest=0.26.6, local=0.26.6
rviz_visual_testing_framework             : latest=14.1.8, local=14.1.7
rviz2                                     : latest=14.1.8, local=14.1.7
rviz_default_plugins                      : latest=14.1.8, local=14.1.7
rviz_common                               : latest=14.1.8, local=14.1.7
rqt_bag                                   : latest=1.5.4, local=1.5.4
rosbag2_examples_py                       : latest=0.26.6, local=0.26.6
launch_testing_examples                   : latest=0.19.5, local=0.19.4
ros2bag                                   : latest=0.26.6, local=0.26.6
rosbag2_py                                : latest=0.26.6, local=0.26.6
rosbag2_transport                         : latest=0.26.6, local=0.26.6
rosbag2_storage_sqlite3                   : latest=0.26.6, local=0.26.6
rosbag2_examples_cpp                      : latest=0.26.6, local=0.26.6
rosbag2_compression                       : latest=0.26.6, local=0.26.6
rosbag2_cpp                               : latest=0.26.6, local=0.26.6
rosbag2_storage                           : latest=0.26.6, local=0.26.6
image_common                              : latest=5.1.6, local=5.1.5
camera_info_manager                       : latest=5.1.6, local=5.1.5
camera_calibration_parsers                : latest=5.1.6, local=5.1.5
yaml_cpp_vendor                           : latest=9.0.1, local=9.0.1
interactive_markers                       : latest=2.5.4, local=2.5.4
common_interfaces                         : latest=5.3.6, local=5.3.5
visualization_msgs                        : latest=5.3.6, local=5.3.5
dummy_robot_bringup                       : latest=0.33.5, local=0.33.5
robot_state_publisher                     : latest=3.3.3, local=3.3.3
kdl_parser                                : latest=2.11.0, local=2.11.0
urdf                                      : latest=2.10.0, local=2.10.0
urdf_parser_plugin                        : latest=2.10.0, local=2.10.0
[... 397 lines cut ...]

### Actual behavior

When QoS::keep_last(0) is called:

No exception is thrown;

A warning log is printed:
```cpp
[WARN] [rclcpp]: A zero depth with KEEP_LAST doesn't make sense; no data could be stored. This will be interpreted as SYSTEM_DEFAULT
```

### Additional information

## Suggested Fix
```cpp
QoS & QoS::keep_last(size_t depth) {
  if (depth == 0) {
    throw std::invalid_argument("keep_last(depth): depth must be greater than 0");
  }
  qos_profile_.history = RMW_QOS_POLICY_HISTORY_KEEP_LAST;
  qos_profile_.depth = depth;
  return *this;
}
```

fujitatomoya, 2025-04-14:
> Currently, when rclcpp::QoS::keep_last(0) is called, the system logs a warning and silently falls back to RMW_QOS_POLICY_HISTORY_SYSTEM_DEFAULT,Currently, when rclcpp::QoS::keep_last(0) is called, the system logs a warning and silently falls back to RMW_QOS_POLICY_HISTORY_SYSTEM_DEFAULT,

it does not silently fall back to SYSTEM_DEFAULT, instead it clearly warns the user application with `[WARN] [1744650571.919250788] [rclcpp]: A zero depth with KEEP_LAST doesn't make sense; no data could be stored.This will be interpreted as SYSTEM_DEFAULT`.

i think this fallback design is implemented as expected, but we can discuss on throwing exception instead. i think that current permissive behavior is fine with warning message.

zhihaoshang, 2025-04-15:
> > 目前，当调用 rclcpp：：QoS：：keep_last（0） 时，系统会记录警告并静默回退到 RMW_QOS_POLICY_HISTORY_SYSTEM_DEFAULT，目前，当调用 rclcpp：：QoS：：keep_last（0） 时，系统会记录警告并静默回退到 RMW_QOS_POLICY_HISTORY_SYSTEM_DEFAULT，
> 
> 它不会静默地回退到 SYSTEM_DEFAULT，而是清楚地使用 .`[WARN] [1744650571.919250788] [rclcpp]: A zero depth with KEEP_LAST doesn't make sense; no data could be stored.This will be interpreted as SYSTEM_DEFAULT`
> 
> 我认为这种回退设计已按预期实现，但我们可以讨论引发异常。我认为当前的宽容行为对警告消息没有问题。

Thank you for your clarification. I agree that the warning is visible at runtime, and the current behavior avoids crashes, which has practical benefits.

However, I believe that explicitly throwing an std::invalid_argument exception for keep_last(0) would provide clearer feedback during development, especially in automated testing scenarios where logs might not be closely monitored.

Failing silently (even with a warning) could lead to misconfigured QoS settings, which are hard to trace and may introduce subtle runtime bugs.

I suggest we consider adopting a stricter policy for input validation in QoS settings to improve overall robustness. This change can even be guarded with a macro or optional compile-time flag to preserve backward compatibility if needed.

I hope the team would consider submitting a patch or PR to address this, as it could help improve the clarity and robustness of QoS configuration handling.

fujitatomoya, 2025-04-15:
@zhihaoshang according to the history, are you using GenAI or any static tools to create the issue? can you share what kind of tools are you using to create issue and response?

Reference: https://discourse.ros.org/t/interim-policy-on-the-use-of-generative-ai-in-osrf-projects/39731

zhihaoshang, 2025-04-16:
> 根据历史记录，您是否使用 GenAI 或任何静态工具来制造问题？您能否分享一下您使用了哪些类型的工具来创建 Issue 和 Response？
> 
> 编号： https://discourse.ros.org/t/interim-policy-on-the-use-of-generative-ai-in-osrf-projects/39731

Hi, thank you for your reply! I used some auxiliary tools when writing the responses, currently using ChatGPT, to help improve the clarity of expression. Please feel free to let me know if there's anything inappropriate.

closed 2025-04-16 by zhihaoshang