ros2/rclcpp #2812: QoSInitialization::from_rmw does not validate invalid history policy values, leading to silent failures

zhihaoshang, 2025-04-14:
### Operating System:

```Linux shangzh-VMware-Virtual-Platform 6.11.0-21-generic #21~24.04.1-Ubuntu SMP PREEMPT_DYNAMIC Mon Feb 24 16:52:15 UTC 2 x86_64 x86_64 x86_64 GNU/Linux```

### ROS version or commit hash:

ros2 jazzy

### Client library (if applicable):

rclcpp

### 'ros2 doctor --report' output

<details><summary>ros2 doc --report</summary>

```console
   NETWORK CONFIGURATION
inet         : 127.0.0.1
inet4        : ['127.0.0.1']
inet6        : ['::1']
netmask      : 255.0.0.0
device       : lo
flags        : 73<RUNNING,UP,LOOPBACK>
mtu          : 65536
inet         : 192.168.148.137
inet4        : ['192.168.148.137']
ether        : 00:0c:29:be:c8:19
inet6        : ['fe80::20c:29ff:febe:c819%ens33']
netmask      : 255.255.255.0
device       : ens33
flags        : 4163<RUNNING,UP,MULTICAST,BROADCAST>
mtu          : 1500
broadcast    : 192.168.148.255

   PACKAGE VERSIONS
rqt_bag_plugins                           : latest=1.5.4, local=1.5.4
rosbag2                                   : latest=0.26.6, local=0.26.6
rosbag2_storage_default_plugins           : latest=0.26.6, local=0.26.6
rosbag2_storage_mcap                      : latest=0.26.6, local=0.26.6
rosbag2_compression_zstd                  : latest=0.26.6, local=0.26.6
mcap_vendor                               : latest=0.26.6, local=0.26.6
zstd_vendor                               : latest=0.26.6, local=0.26.6
rviz_visual_testing_framework             : latest=14.1.8, local=14.1.7
rviz2                                     : latest=14.1.8, local=14.1.7
rviz_default_plugins                      : latest=14.1.8, local=14.1.7
rviz_common                               : latest=14.1.8, local=14.1.7
rqt_bag                                   : latest=1.5.4, local=1.5.4
rosbag2_examples_py                       : latest=0.26.6, local=0.26.6
launch_testing_examples                   : latest=0.19.5, local=0.19.4
ros2bag                                   : latest=0.26.6, local=0.26.6
rosbag2_py                                : latest=0.26.6, local=0.26.6
rosbag2_transport                         : latest=0.26.6, local=0.26.6
rosbag2_storage_sqlite3                   : latest=0.26.6, local=0.26.6
rosbag2_examples_cpp                      : latest=0.26.6, local=0.26.6
rosbag2_compression                       : latest=0.26.6, local=0.26.6
rosbag2_cpp                               : latest=0.26.6, local=0.26.6
rosbag2_storage                           : latest=0.26.6, local=0.26.6
image_common                              : latest=5.1.6, local=5.1.5
camera_info_manager                       : latest=5.1.6, local=5.1.5
camera_calibration_parsers                : latest=5.1.6, local=5.1.5
yaml_cpp_vendor                           : latest=9.0.1, local=9.0.1
interactive_markers                       : latest=2.5.4, local=2.5.4
common_interfaces                         : latest=5.3.6, local=5.3.5
visualization_msgs                        : latest=5.3.6, local=5.3.5
dummy_robot_bringup                       : latest=0.33.5, local=0.33.5
robot_state_publisher                     : latest=3.3.3, local=3.3.3
kdl_parser                                : latest=2.11.0, local=2.11.0
urdf                                      : latest=2.10.0, local=2.10.0
urdf_parser_plugin                        : latest=2.10.0, local=2.10.0
[... 388 lines cut ...]
### Expected behavior

When an invalid history_policy enumeration value is provided (i.e., not one of RMW_QOS_POLICY_HISTORY_KEEP_ALL, KEEP_LAST, SYSTEM_DEFAULT, or UNKNOWN), the function should explicitly throw an exception (e.g., std::invalid_argument) to indicate that the input is invalid and prevent unintended behavior.

### Actual behavior

The function does not check the validity of the history_policy value. Any unsupported value silently falls into the default case and proceeds with KeepLast(rmw_qos.depth), making it impossible for the caller to detect the invalid configuration.

### Additional information

## Suggested Fix
Add explicit validation for the history_policy value. For example, the default case in the switch statement can be modified to throw an exception:
```cpp
switch (rmw_qos.history) {
  case RMW_QOS_POLICY_HISTORY_KEEP_ALL:
    return KeepAll();
  case RMW_QOS_POLICY_HISTORY_SYSTEM_DEFAULT:
    return KeepLast(rmw_qos.depth, false);
  case RMW_QOS_POLICY_HISTORY_KEEP_LAST:
  case RMW_QOS_POLICY_HISTORY_UNKNOWN:
    return KeepLast(rmw_qos.depth);
  default:
    throw std::invalid_argument("Invalid history policy enum value passed to QoSInitialization::from_rmw");
}
```

fujitatomoya, 2025-04-14:
test code generates the following warnings and error during build time.

```console
warning: the result of the conversion is unspecified because ‘-1’ is outside the range of type ‘rmw_qos_history_policy_t’ {aka ‘rmw_qos_history_policy_e’} [-Wconversion]
  369 |   rmw_qos_profile_t rmw_qos_invalid = {static_cast<rmw_qos_history_policy_t>(-1), 0, {}};
      |                                                                              ^~

```

and ,

```console
/root/ros2_ws/colcon_ws/src/ros2/ros2_test_prover/prover_rclcpp/src/build_test.cpp:13:88: error: missing initializer for member ‘rmw_qos_profile_s::durability’ [-Werror=missing-field-initializers]
   13 |   rmw_qos_profile_t rmw_qos_invalid = {static_cast<rmw_qos_history_policy_t>(-1), 0, {}};
      |                                                                                        ^
/root/ros2_ws/colcon_ws/src/ros2/ros2_test_prover/prover_rclcpp/src/build_test.cpp:13:88: error: missing initializer for member ‘rmw_qos_profile_s::deadline’ [-Werror=missing-field-initializers]
/root/ros2_ws/colcon_ws/src/ros2/ros2_test_prover/prover_rclcpp/src/build_test.cpp:13:88: error: missing initializer for member ‘rmw_qos_profile_s::lifespan’ [-Werror=missing-field-initializers]
/root/ros2_ws/colcon_ws/src/ros2/ros2_test_prover/prover_rclcpp/src/build_test.cpp:13:88: error: missing initializer for member ‘rmw_qos_profile_s::liveliness’ [-Werror=missing-field-initializers]
/root/ros2_ws/colcon_ws/src/ros2/ros2_test_prover/prover_rclcpp/src/build_test.cpp:13:88: error: missing initializer for member ‘rmw_qos_profile_s::liveliness_lease_duration’ [-Werror=missing-field-initializers]
/root/ros2_ws/colcon_ws/src/ros2/ros2_test_prover/prover_rclcpp/src/build_test.cpp:13:88: error: missing initializer for member ‘rmw_qos_profile_s::avoid_ros_namespace_conventions’ [-Werror=missing-field-initializers]
```

@zhihaoshang can you provide your reproducible example project instead of code snippet? i am unable to build your test and warning message tells that is out of range clearly.

ahcorde, 2025-05-15:
Related PR https://github.com/ros2/rclcpp/pull/2841