ros2/rclcpp #2817: RingBufferImplementation::clear() does not actually clear buffer contents

zhihaoshang, 2025-04-15:
### Operating System:

```Linux shangzh-VMware-Virtual-Platform 6.11.0-21-generic #21~24.04.1-Ubuntu SMP PREEMPT_DYNAMIC Mon Feb 24 16:52:15 UTC 2 x86_64 x86_64 x86_64 GNU/Linux```

### ROS version or commit hash:

ros2 jazzy

### Client library (if applicable):

rclcpp

### 'ros2 doctor --report' output

<details><summary>ros2 doc --report</summary>

```console
   NETWORK CONFIGURATION
inet         : 127.0.0.1
inet4        : ['127.0.0.1']
inet6        : ['::1']
netmask      : 255.0.0.0
device       : lo
flags        : 73<UP,RUNNING,LOOPBACK>
mtu          : 65536
inet         : 192.168.148.137
inet4        : ['192.168.148.137']
ether        : 00:0c:29:be:c8:19
inet6        : ['fe80::20c:29ff:febe:c819%ens33']
netmask      : 255.255.255.0
device       : ens33
flags        : 4163<UP,MULTICAST,RUNNING,BROADCAST>
mtu          : 1500
broadcast    : 192.168.148.255

   PACKAGE VERSIONS
rqt_bag_plugins                           : latest=1.5.4, local=1.5.4
rosbag2                                   : latest=0.26.6, local=0.26.6
rosbag2_storage_default_plugins           : latest=0.26.6, local=0.26.6
rosbag2_storage_mcap                      : latest=0.26.6, local=0.26.6
rosbag2_compression_zstd                  : latest=0.26.6, local=0.26.6
mcap_vendor                               : latest=0.26.6, local=0.26.6
zstd_vendor                               : latest=0.26.6, local=0.26.6
rviz_visual_testing_framework             : latest=14.1.8, local=14.1.7
rviz2                                     : latest=14.1.8, local=14.1.7
rviz_default_plugins                      : latest=14.1.8, local=14.1.7
rviz_common                               : latest=14.1.8, local=14.1.7
rqt_bag                                   : latest=1.5.4, local=1.5.4
rosbag2_examples_py                       : latest=0.26.6, local=0.26.6
launch_testing_examples                   : latest=0.19.5, local=0.19.4
ros2bag                                   : latest=0.26.6, local=0.26.6
rosbag2_py                                : latest=0.26.6, local=0.26.6
rosbag2_transport                         : latest=0.26.6, local=0.26.6
rosbag2_storage_sqlite3                   : latest=0.26.6, local=0.26.6
rosbag2_examples_cpp                      : latest=0.26.6, local=0.26.6
rosbag2_compression                       : latest=0.26.6, local=0.26.6
rosbag2_cpp                               : latest=0.26.6, local=0.26.6
rosbag2_storage                           : latest=0.26.6, local=0.26.6
image_common                              : latest=5.1.6, local=5.1.5
camera_info_manager                       : latest=5.1.6, local=5.1.5
camera_calibration_parsers                : latest=5.1.6, local=5.1.5
yaml_cpp_vendor                           : latest=9.0.1, local=9.0.1
interactive_markers                       : latest=2.5.4, local=2.5.4
common_interfaces                         : latest=5.3.6, local=5.3.5
visualization_msgs                        : latest=5.3.6, local=5.3.5
dummy_robot_bringup                       : latest=0.33.5, local=0.33.5
robot_state_publisher                     : latest=3.3.3, local=3.3.3
kdl_parser                                : latest=2.11.0, local=2.11.0
urdf                                      : latest=2.10.0, local=2.10.0
urdf_parser_plugin                        : latest=2.10.0, local=2.10.0
[... 409 lines cut ...]
The current implementation of clear() is:
```cpp
void clear() override {
  TRACETOOLS_TRACEPOINT(rclcpp_ring_buffer_clear, static_cast<const void *>(this));
}
```
This implementation does not modify the buffer contents or internal state. After invoking clear(), the buffer still contains data, and has_data() returns true, which is inconsistent with the expected behavior.

### Additional information

## Suggested Fix
Update the implementation of clear() to reset the buffer contents and internal state in a thread-safe manner. For example:
```cpp
void clear() override {
  std::lock_guard<std::mutex> lock(mutex_);
  for (auto & item : ring_buffer_) {
    item.reset();  // Properly release resources if BufferT is a smart pointer
  }
  size_ = 0;
  read_index_ = 0;
  write_index_ = 0;

  TRACETOOLS_TRACEPOINT(rclcpp_ring_buffer_clear, static_cast<const void *>(this));
}
```

fujitatomoya, 2025-04-16:
@zhihaoshang i see you are removing comments and closing issues that you opened recently all the sudden. do you want to keep this one? or close too?

zhihaoshang, 2025-04-16:
> 我看到您突然删除了评论并关闭了您最近打开的问题。你想保留这个吗？还是很接近？

Hello, I closed some issues before because with everyone's explanation and help, I realized that those may be my own understanding biases and not real bugs, so I chose to close them. Thank you very much for your patience. But I haven't found a workaround to this issue yet, and I do need some help, so I hope to keep this issue open.

alsora, 2025-04-25:
@zhihaoshang as you pointed out, the current implementation does nothing.

> But I haven't found a workaround to this issue yet, and I do need some help

What do you mean that you haven't found a workaround? Could you please describe what did you try and what didn't work?
In your post you are providing a "suggested fix" which seems correct at a quick look.
What's the problem with it?

Note that this method is not used anywhere in the codebase.
There's also no risk of resource leak because data is stored via smart pointers which will get released upon destruction.

zhihaoshang, 2025-04-25:
> 正如您所指出的，当前的实现不执行任何作。
> 
> > 但我还没有找到这个问题的解决方法，我确实需要一些帮助
> 
> 您没有找到解决方法是什么意思？您能否描述一下您尝试了什么，什么无效？在您的帖子中，您提供了一个“建议的修复”，乍一看似乎是正确的。它有什么问题？
> 
> 请注意，此方法未在代码库中的任何位置使用。也没有资源泄漏的风险，因为数据是通过智能指针存储的，这些指针将在销毁时被释放。

Thank you for your patience! Since I'm not familiar with submitting PRs yet, I've tried to suggest some fixes that might work. Hopefully, developers who are more familiar with the project can refer to these suggestions and help submit a PR for fixes.

Castleone98, 2025-04-25:
I’d like to work on this fix.  
Let me know if anyone is already tackling it—otherwise I’ll open a PR shortly. 🙂

zhihaoshang, 2025-04-25:
> I’d like to work on this fix. Let me know if anyone is already tackling it—otherwise I’ll open a PR shortly. 🙂

Thank you very much for your reply, no one has done a fix yet, if you can fix it, thank you very much

PR #2837 (2025-05-04): Update RingBufferImplementation to clear internal data.
Closes #2817
closed 2025-05-04 by fujitatomoya