ros2/rclcpp #2818: Adding custom thread names to component_container_isolated

Aposhian, 2025-04-16:
### Description

Since component_container_isolated spins up a thread for each node, it would be nice to name those threads in a way that operating system tools can identify them easily. This would involve writing code that uses the platform-specific API's with conditional compilation that sets the thread name at the beginning.

### Motivation

Identify resource usage quickly by node using tools like htop, system monitor, task manager, etc.

### Design / Implementation Considerations

I think it would be best to use the fully qualified node name, but with namespaces, this could quickly get longer than the character limit for thread names, and would require truncation. So it might be better to do just the node name itself, or maybe some kind of identifier, maybe just an index, that could be logged with debug logs?

fujitatomoya, 2025-04-17:
it could be useful for frontend monitoring tool, but what if `ComponentManagerIsolated` is templated with `rclcpp::executors::MultiThreadedExecutor`? any thoughts?

Aposhian, 2025-04-17:
> it could be useful for frontend monitoring tool, but what if `ComponentManagerIsolated` is templated with `rclcpp::executors::MultiThreadedExecutor`? any thoughts?

I suppose even if you are using multi threaded executor, each component will have an isolated thread pool, correct? So it could still be helpful to just have all of the threads in a components pool be named after the node.

fujitatomoya, 2025-04-18:
> each component will have an isolated thread pool, correct?

yes, but we need to update the MultiThreadedExecutor class to allow the application to change the names of all threads of it. we can control the main thread, but not all child threads created by MultiThreadedExecutor class.

> So it could still be helpful to just have all of the threads in a components pool be named after the node.

are you suggesting that all the child thread in the MultiThreadedExecutor should inherit the same name from main thread of that?

Aposhian, 2025-04-18:
> are you suggesting that all the child thread in the MultiThreadedExecutor should inherit the same name from main thread of that?

I think so, because then at least different components with their own executors would have differently named threads from each other.

MichaelOrlov, 2025-05-01:
At a weekly ROS 2 issues triage meeting, we agreed that yes, it makes sense to add it. However, help wanted here with implementation.

Aposhian, 2025-05-01:
I can take a try at this.

ahcorde, 2025-06-19:
Thank you for your contribution @Aposhian

PR #2871 (2025-06-18): set thread names by node in component container isolated
## Description
Set the thread name of each node's thread in `component_container_isolated`. This should support POSIX platforms, Apple, and Windows. I tested on Linux only so far.

Fixes #2818

### Is this user-facing behavior change?
User's may see an additional warning if the thread name couldn't set for whatever reason. If users are looking at custom thread names, then they will see a difference.

### Did you use Generative AI?
Yes with Cursor autocomplete.

### Additional Information