ros2/rclcpp #2820: Potential thread-safe issue on wait and guard condition

evshary, 2025-04-18:
### Operating System:

Ubuntu 22.04, 24.04

### ROS version or commit hash:

Humble, Rolling

### RMW implementation (if applicable):

rmw_zenoh_cpp, rmw_cyclonedds_cpp, rmw_fastrtps_cpp

### RMW Configuration (if applicable):

N/A

### Client library (if applicable):

rclcpp

### 'ros2 doctor --report' output

N/A

### Steps to reproduce issue

1. Build the code here: https://github.com/evshary/ros2-simple-examples/blob/e2a7ae145bffbbe71654512bc51db4e48c4a8182/simple_examples_cpp/src/timer/timer.cpp
2. Run `ros2 run simple_examples_cpp timer` 

### Expected behavior

Should keep working

### Actual behavior

The timer will stop

### Additional information

As we summarized at https://github.com/ros2/rmw_zenoh/issues/608#issuecomment-2812540977
The issue comes from `rmw_destroy_guard_condition` being called during `rmw_wait`.
`rmw_wait` will access an invalid memory address and cause an undefined behavior issue.

For the humble, it's much easier to reproduce the issue. Although it greatly improves after the rolling version, it might still happen in the long run.

Since it happens in [different DDS implementations](https://github.com/ros2/rmw_zenoh/issues/608#issuecomment-2814394501), we decided to open an issue here.

jmachowinski, 2025-04-18:
Which version of rolling did you exactly use to test this ?

FYI Normally the executor **should** hold strong reference to all entities in use during the rcl_wait (which call rmw_wait). Therefore dropping a shared ptr to a entity or a callback group in a callback **should** during an ongoing wait should not result in a real deletion until the wait has finished. However there have been bug around this area in the past. Some of them have been fixed about 2 month ago.

I'll check if I can reproduce this with the latest rolling version. 
Note, I will not check / fix humble.

jmachowinski, 2025-04-18:
I tried to reproduce this issue on rolling and had no luck so far. 
Valgrind seems to be happy, and hellgrind is all over the place, but most stuff seems to originate either from the test case itself (non synchronized write to std::cout) or fastDDS.
I also went over the code and don't see how this could happen. All guard conditions in this test case should originate from the callback groups, and are added to the ExecutorNotifyWaitable. The ExecutorNotifyWaitable takes hard ownership in the latest version of rolling of them. The ExecutorNotifyWaitable in use is added to the collection as a Waitable.
Before the wait all waitables are referenced hard as shared_ptr, and released after the wait.

@evshary Sorry, but I will need to have a better reproducer for rolling in order to pursue this further.

evshary, 2025-04-20:
Hi @jmachowinski 
Thank you for the detailed investigation. The rolling version I used is 2025.03. I found that there is an update in 2025.04.
After updating the packages, the issue disappeared. The fix you mentioned is probably included in this update.
If the patch will not be backported to humble, I think we can close the issue.
Thank you for the help!

closed 2025-04-20 by evshary

jmachowinski, 2025-04-22:
The fix from rolling is not back portable. 
Humble is still open for bug fixes though. (I just said, that **I** won't touch humble...)
Feel free to dig deep into the executor code and fix the issue for good ;-)