ros2/rclcpp #2826: Backport Request: PR #2617 (Implement generic service) for Jazzy

TarikViehmann, 2025-04-25:
As jazzy is the latest LTS version, it would be nice to have the work done to generic services backported as without it it feels incomplete to only offer generic client but not service support.
The relevant PR https://github.com/ros2/rclcpp/pull/2617 has (as far as I can tell) no breaking ABI or API changes and (seemingly) does not require anything from rmw that is not already there.
Hence backporting should be doable I think (did so locally on my machine, there it compiles, but I did not actually use it yet).

Related to this, https://github.com/ros2/rclcpp/pull/2662 can also be backported (just fixes an erroneous comment).

There is also this PR https://github.com/ros2/rclcpp/pull/2614, but sadly it breaks ABI by extending an existing variant, so I guess this would be out of reach (but tbh, It is more of a QoL thing and I think jazzy would greatly benefit from my main request, even without this).

Thank you all for working on the generic introspection support, it is really awesome for folks like me that build generic interfaces from foreign frameworks to ROS.

alsora, 2025-05-09:
@fujitatomoya can you take a look?
It's possible that this is not back-portable

fujitatomoya, 2025-05-09:
I think https://github.com/ros2/rclcpp/pull/2617 and https://github.com/ros2/rclcpp/pull/2662 are API/ABI compatible so that we can backport but https://github.com/ros2/rclcpp/pull/2614.

but we do not have any plan to do that, so if someone can work on the backport (code base could be different so might need to resolve the conflict and tests) that is fine, i think.