ros2/rclcpp #2832: ROS2 Action Server getting info of the client node name and the priority

saikishor, 2025-04-30:
### Description

Hello!

Thank you all for such great work. Most of the times in real applications, the action server should show different behaviour depending on the type of the node, for instance, an emergency movement in joints should prevail any motion it is executing or any motion it will receive in future. However, there is no clear line of priority setting in the goals, and i believe it shouldn't be particular to a specific action. If not, most of the released packages needed to be modified inorder to handle this kind of functionality.

Regarding the node name, it would be interesting for the action server to know from whom it is receiving the goal, this information can be used by the server to accept or reject goals. For instance, I want only authorized applications be able to send goals to the hardware and not anyone, inorder to avoid any uncontrolled movements

### Motivation

The motivation is for the hardware based ros2 controllers the action server will  need to handle the resource sharing and this way it can give critical applications more preference than any kind of application sending random movement (or) a robot performing human robot interaction and then it being preempted as navigation needs the head of the robot. These kind of cases are pretty usual in any application.

I would like to hear your views.

Thank you

fujitatomoya, 2025-04-30:
@saikishor thanks for opening the discussion, i think this is interesting.

> However, there is no clear line of priority setting in the goals, and i believe it shouldn't be particular to a specific action. If not, most of the released packages needed to be modified inorder to handle this kind of functionality.

i think it is reasonable that action server prioritize the goal request from the multiple clients, so that high priority goal request prevails to low priority goal requests. and i can see your use case as well. but i am not sure if we can do that with current design of actions. basically all goal requests are stored in rmw implementation queue, and once it is ready ROS 2 takes out the data and open the package to do some work. that said, for doing this, ROS 2 needs to take out the all data once they are ready before executing, and cache them in the ROS 2 action queue to determine what goal request needs to be done 1st with priority. besides this, how to execute the goal is up to the application after accepted callback is called. and i think this makes sense to pass the goal request as fast as possible to the user application and the application decides how to process it.

another approach here could be like `PrioritizedActionServer` in the application layer? having the manager thread to be called by accepted callback, and also manages the thread pools so that it dispatches the goal to be executed based on the priority in the goal (user defined) request message?

IMO, i would love to see this feature in more generic functionality like `Per-Message Priority` in rmw. the priority use case in the same entity is common not only for action but service and topic. (for example, the application wants to receive the priority message 1st than anything else if those are already in rmw queue.) that said, if ROS 2 can take out the data from rmw implementation in prioritized order as multiple senders send so, current system just works without any extra cache mechanism.

i just did a quick brainstorming by myself, i may be wrong at some part. i would love to hear more feedbacks on this idea.

saikishor, 2025-04-30:
Thank you so much @fujitatomoya  for your prompt response. 

> i think it is reasonable that action server prioritize the goal request from the multiple clients, so that high priority goal request prevails to low priority goal requests. and i can see your use [...]

Interesting, in this approach, how do we inject the priority of the goal? Similar approach of how the GoalUUID is generated by the clients, we add a method to set the GoalPriority? 

> ROS 2 needs to take out the all data once they are ready before executing, and cache them in the ROS 2 action queue to determine what goal request needs to be done 1st with priority. besides this, [...]

If I understand your correctly, instead of receiving goal by goal in the `GoalCallback` goal handle callback, receive all of them and the action server decides what to do with each of them?

> another approach here could be like `PrioritizedActionServer` in the application layer? having the manager thread to be called by accepted callback, and also manages the thread pools so that it dispatches the goal to be executed based on the priority in the goal (user defined) request message?

It would totally make sense to have it handled at the lower level instead of another new server.  

> IMO, i would love to see this feature in more generic functionality like `Per-Message Priority` in rmw. the priority use case in the same entity is common not only for action but service and topic. [...]

I agree with you on this part, it would be easier if ROS 2 could take out the data directly. 

@fujitatomoya, What do you think about having access to the information on who sent the goal available to the action servers? In some instances, we might want the action goals to be received only by the authorized applications rather than by anyone.

fujitatomoya, 2025-05-01:
> What do you think about having access to the information on who sent the goal available to the action servers? In some instances, we might want the action goals to be received only by the authorized applications rather than by anyone.

i think this is where https://github.com/ros2/sros2 comes to play. that enables us to discover the only authenticated node(context) and authorized endpoint with access control. that said, anybody else who does not have appropriate security enclaves, they will not see the ROS 2 network at all.