ros2/rclcpp #2842: Replace std::default_random_engine with std::mt19937 in client_base.cpp

keeponoiro, 2025-05-16:
Currently, std::default_random_engine is used in rclcpp_action/src/client_base.cpp. However, in GCC, this is implemented as minstd_rand0, which is a linear congruential generator (LCG). Due to its periodic nature, when actions are executed frequently, duplicate goal_id values can occur.

This results in the following error in rcl_action_accept_new_goal():

`if (rcl_action_server_goal_exists(action_server, goal_info)) {
    RCL_SET_ERROR_MSG("goal ID already exists");
    return NULL;
}
`

In ServerBase::execute_goal_request_received, this leads to an exception being thrown:
`{
    std::lock_guard<std::recursive_mutex> lock(pimpl_->action_server_reentrant_mutex_);
    rcl_handle = rcl_action_accept_new_goal(pimpl_->action_server_.get(), &goal_info);
}
if (!rcl_handle) {
    throw std::runtime_error("Failed to accept new goal\n");
}
`

Proposed Change
Replacing std::default_random_engine with std::mt19937 in client_base.cpp will eliminate the periodicity issue, ensuring that goal_id values do not duplicate.

Expected Benefits
Prevents duplicate goal_id collisions in frequently executed actions.

Improves the randomness quality of generated IDs.

Avoids unexpected exceptions related to duplicate goal acceptance.

jmachowinski, 2025-05-16:
How many goals does your application generate per hour ? We generate about 4000-6000 actions per hour and never ran into this problem on our production machines.

keeponoiro, 2025-05-16:
@jmachowinski 
My application generates approximately 15,000 actions per hour.

keeponoiro, 2025-05-16:
@jmachowinski 
ros2 humble patch11.

jmachowinski, 2025-05-16:
Feel free to open a pull request. Note, the pull request may not break ABI. 
As the issue does exist any more in newer versions, a fix in rolling first is not required.

keeponoiro, 2025-05-16:
@jmachowinski 
Indeed, in the rolling distribution, goal ID duplication no longer causes exceptions. However, when a duplicate goal ID occurs, it can lead to unintended side effects, such as the expiration time of the first goal being extended.

To ensure more consistent and reliable behavior, it would be beneficial to standardize random number generation using std::mt19937, as it eliminates periodicity and reduces the likelihood of unintended goal ID collisions.

fujitatomoya, 2025-05-16:
> By default, goal_id is retained for 15 minutes

@keeponoiro just FYI, this has been changed into 10 seconds in default. you are using humble, we keep it 15mins with humble.

closed 2025-05-30 by fujitatomoya