ros2/rclcpp #2860: `rclcpp` brings in `lttng-usr` threads by default

hidmic, 2025-05-26:
This isn't quite a bug report nor a feature request. We've noticed that in later ROS 2 distributions, `rclcpp` links in [`tracetools`](https://github.com/ros2/ros2_tracing) by default, which in turn links `lttng-usr`. `lttng-usr` has this fun habit of [spawning threads](https://github.com/lttng/lttng-ust/blob/611ccd9f2f6581faa48b47881520069c30c31daa/src/lib/lttng-ust/lttng-ust-comm.c#L2262-L2270) on `dl` load. That can be problematic when Linux namespaces and [unsharing](https://man7.org/linux/man-pages/man2/unshare.2.html) are in the mix.

Naively, I would imagine that only executables that you want to trace should link `lttng-usr`. `rclcpp` right now makes it so that every ROS 2 aware C++ executable will bring the `lttng` machinery with it.

Could we do better than that? Do we absolutely need to link `lttng-usr` to compile `rclcpp`?

jmachowinski, 2025-05-26:
@christophebedard I guess you might know the answer to this ?

fujitatomoya, 2025-05-26:
> Do we absolutely need to link lttng-usr to compile rclcpp?

IMO. the answer is No.

rclcpp does not fundamentally require lttng-ust to function. It only needs it if you want to use tracing features.
The unconditional dependency is a build system/configuration choice, not a technical necessity.

i can think of 2 possible approaches.

- Make tracetools (and thus lttng-ust) an optional dependency of rclcpp
  - Use CMake options to enable/disable tracing support.
  - Only link tracetools (and lttng-ust) if tracing is enabled.
- Use weak symbols or plugin mechanisms
  - Dynamically load tracing libraries only if tracing is actually requested at runtime.

hidmic, 2025-06-09:
Thanks for the explanation @christophebedard!

> I think having an application-level build-time/runtime switch would be great.

Considering that tracing only goes as far down as `rcl`, I'd argue that opt-out (or opt-in) setup on `rclcpp::init()`, even before `rmw` initialization (which is not traced IIUC) should readily cover the system initialization phase that Section IV-A talks about. If that is not enough, if we want to trace all the way up to `_start()`, we could have a library `__constructor__` ourselves to gate `lttng` loading.

> I personally would strongly prefer making it opt-out rather than opt-in for the reasons I mentioned above.

I don't see why you wouldn't be able to propagate an environment variable across an entire `launch` to opt-in to tracing, but I won't die on that hill. I understand we don't want any regressions for people that rely on tracing.  

Let me draft a patch. It'll be easier to discuss over code.

christophebedard, 2025-06-12:
> Considering that tracing only goes as far down as `rcl`, I'd argue that opt-out (or opt-in) setup on `rclcpp::init()`, even before `rmw` initialization (which is not traced IIUC) should readily [...]

It actually does go down to the `rmw` implementations, e.g., [`rmw_publisher_init`](https://github.com/ros2/ros2_tracing/blob/2ca74e7d891474a470f7228b5b72d40f7e401608/tracetools/include/tracetools/tracetools.h#L182), `rmw_publish`, `rmw_take`, etc.

However, the `rcl_init` tracepoint (called by `rcl_init()`) is indeed probably the first tracepoint called, although a tracepoint in `rmw_init()` would be triggered first (if there was one).

One concern I have is how this would behave if the user application is instrumented with custom, non `ros2:*` tracepoints.

closed 2025-07-31 by christophebedard