ros2/rclcpp #2876: Setting publisher or subscription qos_overriding_options in create_publisher or create_subscription from within a dynamic parameter callback creates a deadlock condition

SteveMacenski, 2025-06-20:
### Generated by Generative AI

N/A

### Operating System:

Ubuntu 24.04

### ROS version or commit hash:

Rolling

### RMW implementation (if applicable):

N/A

### RMW Configuration (if applicable):

N/A

### Client library (if applicable):

rclcpp

### 'ros2 doctor --report' output

N/A

### Steps to reproduce issue

We use `create_publisher` for creating publishers generally and I have been working to expose more of the advanced features within Nav2's LifecycleNode. Including among them `qos_overriding_options` for overriding the QoS policies based on a single parameter `allow_parameter_qos_overrides`.

I found that generally this works fine and there is no issue. However, when using `create_publisher` in the callback of a dynamic parameter to reinitialize an object, the program deadlocks at `rclcpp::create_publisher`. I suspect [around here](https://github.com/ros2/rclcpp/blob/1e95407bdf7d29dce9153591307a7e5380a46c71/rclcpp/include/rclcpp/create_publisher.hpp#L57-L62).

What's interesting is that in all cases, these publishers previously existed so the parameters should have already been declared that this logic is declare or getting. That makes me think that (a) its declaring again and that causes a problem or (b) the `get` logic is wrong and this is being uncovered by running this 2x. 

### Expected behavior

Publisher is created and program can continue as normal

### Actual behavior

Publisher is created and program is permanently deadlocked

SteveMacenski, 2025-06-20:
This is an example: https://github.com/ros-navigation/navigation2/blob/main/nav2_smac_planner/src/smac_planner_hybrid.cpp where as we create a new costmap publisher on `configure` lifecycle state and again in the dynamic parameter callback if required. If the qos overriding options is non-empty, then it deadlocks.

SteveMacenski, 2025-06-23:
Related:
- https://github.com/ros2/rclcpp/issues/2741
- https://github.com/ros2/rclcpp/pull/2742

But I can still see this with the most recent binaries of rclcpp `29.6.0` for publishers. Interestingly enough https://github.com/ros-navigation/navigation2/pull/4968 works now which was the Nav2 PR that brought up the tickets above, but that uses a subscription.

SteveMacenski, 2025-06-24:
I added some logging and it was the mutex because of the deadlock. Basically:
- the `mutex` was locked on Call1 -> new interface was made and hung at `create_publisher`
- Then Call2 came in -> deadlocked because it could not obtain the lock, but it the program was already hung at create_publisher which created the issue

Call2 I think is due to the creation of the publisher, triggering another parameter callback which causes the deadlock on accessing the mutex. I can't reproduce with a subscription. 
 
The hang happens from this path when any of the QoS overrides are enabled
- https://github.com/ros-navigation/navigation2/blob/main/nav2_smac_planner/src/smac_planner_hybrid.cpp#L836
- https://github.com/ros-navigation/navigation2/blob/main/nav2_smac_planner/src/costmap_downsampler.cpp#L54 (`node.expired()` is `false`) 
- https://github.com/ros-navigation/navigation2/blob/main/nav2_costmap_2d/src/costmap_2d_publisher.cpp#L71-L80

I see a second call that then deadlocks trying to obtain the mutex (https://github.com/ros-navigation/navigation2/blob/main/nav2_smac_planner/src/smac_planner_hybrid.cpp#L615) that they other hung call is hogging. 

I made a reproduction example and can't seem to reproduce it with https://github.com/open-navigation/opennav_rclcpp_issue_reproduction. I do see however if I remove those publishers, then everything works properly. I don't think any kind of odd weak pointer issue is at play here, it seems pretty cut and dry from the logging that it hangs creating the publisher triggeres another parameter update for some reason that causes the deadlock. I just can't figure out how...

SteveMacenski, 2025-06-24:
OK. I found root cause @fujitatomoya 

The sub / pubs work fine for creating the interfaces again in the dynamic callbacks **as long as** those interfaces were already existing in the node in the first place. If they're new, and therefore need to declare, that's where the hiccup is.

In the Smac Planner, we only create the object that is hanging on us if the initial settings say so ([example](https://github.com/ros-navigation/navigation2/blob/main/nav2_smac_planner/src/smac_planner_hybrid.cpp#L284-L289)), so when we create it in the dynamic configuration callback, that is the first time it is ever created which causes the hang because its declaring the parameter (I assume).  When I make this object always created ([as in here](https://github.com/ros-navigation/navigation2/pull/5302/files#diff-64c79d55e8fff0f36e4fc702b13a7e62a2963b84870d7e536faca90b6f127f9dR284)) then it works fine.

I was able to show this by adding an arbitrary publisher into the dynamic parameters callback of any node and see the same behavior. I also shows this happens **for subscriptions as well.**

So, the issue is that when a Pub/Sub that enables parameter overrides is created for the first time in a dynamic parameter callback, it hangs when the resource needs to be locked because it triggers a call to the dynamic parameters callback about the declaration.

fujitatomoya, 2025-06-25:
@SteveMacenski thanks for the information.

1st thing as work-around for your application, the resource needs to be locked can use recursive mutex lock? this still makes sure that the resource is locked during multiple parameter callback chain. or i would check the what parameter is being changed in the callback 1st, and then lock the resource if necessary.

i believe that the above is work-around and that is actually a burden for the user application.
the question that i have is that, do we really need to call the user callback when qos override parameters are being declared? probably ordinary parameter requests to those qos override parameters need to be monitored by the user callback, because user callback would watch those qos setting in the callback to accept or reject the incoming requests. but i am not sure if we need to call the user callback on declaration when it creates the publisher and subscription endpoints. this could avoid the callback chain issue reported here?

MarcoMatteoBassa, 2025-09-24:
Hi, we could reproduce an issue (on Jazzy, 28.1.12) that we think is related to the reported one: In Nav2, if enabling the qos_overriding_options for our costmaps, we experience seldom (never occurs on some machines, happens ~10% of times on fast PCs) cases where calls to rclcpp::create_subscription from within another callback, results in a deadlock.

Here an example of filter where this can happen:

https://github.com/ros-navigation/navigation2/blob/c95673edb8be7f38ed2e98f74491c47858d1a82f/nav2_costmap_2d/plugins/costmap_filters/keepout_filter.cpp#L128-L136

Disabling the qos override options completely solves the problem for us.
We verified that the code hangs on the call to "rclcpp::create_subscription" and not while declaring a parameter.

I tried to create a minimal example to reproduce the issue by creating a lot of subscriptions from within a callback, while enabling the qos override options, but I wasn't able to, so unfortunately I am not 100% sure about the easiest path to reproduce.